sparkfun/Fingerprint_Scanner-TTL #13: Verify1_1 function always returns true (even for wrong finger)

fourstix, 2015-01-03:
Hi, 

I have the 5V TTL Fingerprint Scanner  (GT-511C1R) from Sparkfun.  I wrote a demo sketch for my Arduino to exercise all the functions.  I then spent some time playing with the scanner, enrolling several fingers. I noticed that the Verify1_1 function in the library would always return true, even if I gave it the wrong finger, or I gave it an unused ID to verify. 

First I did a hack to the Verify1_1 function and set retval = 3, whenever ACK was false.  I did this in the error code block before it goes through the checks for the error code.  This way the function would always return 3, for Verified Failed, even if the error code wasn't identified.

Next, I looked at the hex codes when verifying the wrong finger.  I see the following debug when I try to verify ID 6 with the wrong finger:

FPS - Verify1_1
FPS - SEND: "55 AA 01 00 06 00 00 00 50 00 56 01"
FPS - RECV: "55 AA 01 00 07 10 00 00 31 00 48 01"

The high byte of the error code is 0x10 not 0x01 for my scanner.  The error code is 0x07 (Verified Failed) as expected for the wrong finger.   I made a change to the logic in 
Response_Packet::ErrorCodes::Errors_Enum Response_Packet::ErrorCodes::ParseFromBytes(byte high, byte low) logic
to change the if (high == 0x01) clause to an else clause, for any non-zero high byte.  This will cause the code to work correctly for my scanner.  I wasn't sure if the 3.3v versions return a high-byte of 0x01 or not, so I thought I should code for multiple possible non-zero values.

I branched the code and I will make the changes in my own branch.  You're welcome to go and look at the changes and let me know if the changes I made are valid. 

Thanks for your library. I've had a lot of fun playing with it.

Best wishes,
Gaston Williams

MH6, 2015-01-27:
Yep, had the same issue with error codes not being parsed correctly on my GT-511C3. Had to modify Response_Packet::ErrorCodes::ParseFromBytes and add a check for 0x10 as well.

JoshHawley, 2015-06-18:
Those changes look good to me. Thanks for your help =)

ToniCorinne, 2015-06-19:
Hey @fourstix - to ensure I don't pull in the wrong branch from your repo, please submit a pull request with the specific patch you are referring to. We can then get those updates added (and make sure you're given credit!).

bboyho, 2017-09-25:
Hmmm, I think I found the original pull request which is issued in Josh Hawley's repository waaay back in time => [https://github.com/JoshHawley/Fingerprint_Scanner-TTL/pull/1](https://github.com/JoshHawley/Fingerprint_Scanner-TTL/pull/1) .

bboyho, 2018-03-15:
It looks like your fix was merged waaaay back in time [ [https://github.com/JoshHawley/Fingerprint_Scanner-TTL/pull/1](https://github.com/JoshHawley/Fingerprint_Scanner-TTL/pull/1) ]. I will be closing this issue. Thank you sharing the fix with the community! =)

closed 2018-03-15 by bboyho