sparkfun/Fingerprint_Scanner-TTL #14: Error Compiling

aizirt, 2015-01-21:
Hi, I've tried compiling but I always get these errors:

FPS_GT511C3\FPS_GT511C3.cpp.o: In function `Command_Packet::GetPacketBytes()':
FPS_GT511C3.cpp:17: undefined reference to `operator new[](unsigned int)'
FPS_GT511C3\FPS_GT511C3.cpp.o: In function `FPS_GT511C3::GetResponse()':
FPS_GT511C3.cpp:735: undefined reference to `operator new[](unsigned int)'

aizirt, 2015-01-23:
FPS_GT511C3\FPS_GT511C3.cpp.o: In function `Command_Packet::GetPacketBytes()':
C:\Users\Steven Magtibay\Documents\New folder\arduino-1.0.6\libraries\FPS_GT511C3/FPS_GT511C3.cpp:17: undefined reference to `operator new[](unsigned int)'
FPS_GT511C3\FPS_GT511C3.cpp.o: In function `FPS_GT511C3::GetResponse()':
C:\Users\Steven Magtibay\Documents\New folder\arduino-1.0.6\libraries\FPS_GT511C3/FPS_GT511C3.cpp:735: undefined reference to `operator new[](unsigned int)'

eduardonunes13, 2015-02-28:
did you have your problem solved? i have the same....but i'm still not found the answer

JoshHawley, 2015-06-18:
This looks like a problem with the compiler/linker to me. new is a pretty basic keyword..

closed 2015-06-19 by ToniCorinne

abluke, 2015-09-10:
@eduardonunes13 @aizirt Have you guys found the solution for this problem?
I am stuck with this error. Please help me out if you guys have sorted it. :)

Thanks,
Abel Luke