sparkfun/Fingerprint_Scanner-TTL #15: handling more than 200 fingerprints

thiagocaproni, 2015-03-19:


JoshHawley, 2015-06-18:
The fingerprint scanner has storage for 200 fingerprints. Downloading the fingerprint templates from the FPS is possible, but downloading them to an arduino is really difficult because it does not have the memory needed to store them (even temporarily). While I'm sure that it is possible to stream the file through the arduino out to the network or other storage, it is fairly complex code that really extends beyond this library's scope.

JoshHawley, 2015-06-18:
All that said, I do plan on looking at this issue again in the future. It is something that I want to solve so that I can make multiple FPS scanners update their local templates from a central repository.

closed 2015-06-19 by ToniCorinne

ahsunali, 2017-08-29:
Hi, we are experiencing the same issue, as we want to use the scanner to scan the fingerprints and store the templates in a Database , in fact the templates definitely exceeds the 200 limit, whats should be the work arround in this case ?? 
Thanks