sparkfun/Fingerprint_Scanner-TTL #17: Functions "GetTemplate()" and "SetTemplate()" doesn't work on arduino.

rlepricon, 2015-06-30:
Hi,

Functions "GetTemplate()" and "SetTemplate()" doesn't work on arduino or I can't understand how to use them.

JoshHawley, 2015-09-21:
They are not implemented. There are comments as placeholders for them, but that is all. They were not implemented due to memory restrictions on the Arduino.

IamNito, 2016-01-17:
Could I just use the serial connection to use those functions? As far as I get it, the sensor could understand that somehow.. [section 5.19. Make Template(MakeTemplate) in the data sheet]
I really need that function in my project because I cannot save any templates on the chip due to security requirements. I'll have to do the "comparision" elsewhere, so the template has to get there somehow..
Apart from that it's a great library, @JoshHawley :+1:

felipefuller, 2016-07-11:
somebody got any info about this functions? thanks!

Alarde, 2016-07-11:
I'm really interested on it too.

closed 2019-06-01 by bboyho