sparkfun/Fingerprint_Scanner-TTL #18: Doesn't work well with Microduino Core

abluke, 2015-08-03:
It works well with Arduino Boards. But, I am using Microduino Core+.which is Arduino supported.
When I select the board as Microduino Core+, it shows the following error:

Arduino: 1.6.5 (Windows 7), Board: "Microduino Core+ (644pa), Atmega644pa@8M,3.3V"

Fingerprint_Scanner-TTL-master\FPS_GT511C3.cpp.o: In function `Command_Packet::GetPacketBytes()':
C:\Users\Abel\Documents\Arduino\libraries\Fingerprint_Scanner-TTL-master/FPS_GT511C3.cpp:17: undefined reference to `operator new[](unsigned int)'
Fingerprint_Scanner-TTL-master\FPS_GT511C3.cpp.o: In function `FPS_GT511C3::GetResponse()':
C:\Users\Abel\Documents\Arduino\libraries\Fingerprint_Scanner-TTL-master/FPS_GT511C3.cpp:738: undefined reference to `operator new[](unsigned int)'
collect2.exe: error: ld returned 1 exit status
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
 
Please help me out.

Thanks,
Abel Luke

ToniCorinne, 2015-08-04:
Hi Abel - it looks like the issue might be coming from libraries not being included or linked correctly when compiled in the Arduino IDE. I'd recommend checking the verbose output when compiled for your board and see if that gives you any more information as to where to look. You might also try posting to the [Microduino forum](https://www.microduino.cc/forum/). They should be much more familiar with the build process for that particular board and probably can point out exactly where the issue is.

abluke, 2015-09-08:
Thanks Toni.
I think everything is pretty fine here. As in the compiler as well as the board doesn't have any problem.
I know new is a basic keyword and it should therefore work fine. Also i found same problems with others as well (in some of the closed issues). Is it the problem with the Arduino IDE version that i am using? If yes, which version should i be using here?
Right now i am using Arduino IDE 1.6.5.

Once again,
Thanks,
Abel Luke

ToniCorinne, 2015-09-08:
Hey Abel - My guess is still that something is not nested properly for the build of that particular core. Unfortunately, I'm not familiar with the Microduino core as we don't work with that here, so again, I'd recommend checking the forums for solutions and work-arounds other users have come up with for that particular board. 

You could try running the code in a different version of Arduino, but backwards compatibility could be an issue with the board definition files (again though, I'm not familiar with that platform, so please check the main website for proper information).

abluke, 2015-09-09:
Okay. I will let you know if I sort it out.
Thank you so much.

abluke, 2015-12-09:
Please help me solve this issue. I am not able to move forward without finding a solution.
@ToniCorinne If you can please give it some time and sort out this problem. I am pretty sure there a many who are struggling with this issue because i saw similar issues in your closed issues as well.

ToniCorinne, 2015-12-09:
Different boards in the IDE can require different versions of the libraries. For example, for AVR boards in Arduino, you have [this list of libraries](https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/libraries) that are written for AVRs. However, for SAM boards, there is [this list of libraries](https://github.com/arduino/Arduino/tree/master/hardware/arduino/sam/libraries). Both have an SPI library, but the code is not the same between the two. It varies based on the architecture. The Microduino has its own set of board-specific libraries [here](https://github.com/wasdpkj/Microduino-IDE-win/tree/master/hardware/Microduino/avr/libraries).

The code issue you're having is that the library needs to be modified to be compatible with the Microduino core, but again, we don't work with the Microduino core here at SparkFun. We don't have testing procedures or anyone familiar with that core, and therefore cannot offer support for a board we don't sell. Again, I would recommend checking the Microduino forums as they should have suggestions or pointers on how to port libraries for their core.

bboyho, 2017-08-28:
Hi abluke,

Hawley's Arduino library was written on the Atmega328P with the Arduino Uno bootloader. The FPS_GT511C3 library may not work for all microcontrollers using the Arduino IDE. As you move away from the ATmega328P family, you may need to modify the code or port the library over to get it working. Unfortunately, the FPS_GT511C3 library is not supported for the ATmega644PA on the Microduino Core+.

I recommend using an Arduino with Atmega328P . For something with that size, you might want to look at the 3.3V/8MHz Arduino Pro Mini and a 3.3V FTDI to program.

closed 2017-08-28 by bboyho