sparkfun/Fingerprint_Scanner-TTL #19: Issue with Arduino uno: reply[idx] = Serial.read();

reyli564, 2015-09-29:
Why is it not working? Can you please help me?

JoshHawley, 2015-11-05:
This is not even close to enough information to help you.

bboyho, 2017-08-28:
Hi @reyli564 ,

I agree with Hawley. Your request has no information on your setup (Arduino IDE version, fingerprint scanner model) and what is going on (Arduino compile errors). More information would be necessary to try and troubleshoot the issue that you are having. I tried doing a search in this repository but there are no instances that use "reply[idx] = Serial.read();". Make sure that you are using the GT-511C3 or GT-511C1R fingerprint scanner. There are other fingerprint scanners out in the world and this library does not cover all of them. Doing a search with an online search engine, this looks like the closest GitHut repository used for a different fingerprint scanner [ [https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library/blob/5be04ea14554131325f18f02c7d27ab6532b2075/Adafruit_Fingerprint.cpp#L288](https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library/blob/5be04ea14554131325f18f02c7d27ab6532b2075/Adafruit_Fingerprint.cpp#L288) ]. Based on this information, I do not believe you are using the correct fingerprint scanner model or code.

closed 2017-08-28 by bboyho