sparkfun/Fingerprint_Scanner-TTL #20: Literal string into progmem

gion86, 2015-11-04:
Hello,

I've used  the library for the FPS GT511C1R sensor (but the same holds true for this library), with a small ATTiny84 and I had to comment out the literal string of the serial output for debugging.
The flash memory on the micro is just 8Kb and I had other libraries, so the whole program couldn't fit on the CPU. 
I think these strings should be saved into progmem, and then one can use this functions to read/write to progmem:
http://www.nongnu.org/avr-libc/user-manual/group__avr__pgmspace.html 

Do you think it can work?
Thank you!

JoshHawley, 2015-11-05:
If you are referring to the lines that start with "if (UseSerialDebug)", yes the library will work without them. They are only there to allow an extra stream of data to help debug issues. They are not needed for normal operation.

gion86, 2015-11-05:
I was talking about the string literal that is beyond the UseSerialDebug, like "FPS - Open".
If you move those strings to progmem you can save a lot of flash memory, that is where the actual code goes on the micro.

I was wondering if there is any issue about that? I don't see any..

JoshHawley, 2015-11-05:
Sure if you really want them, but I was suggesting that you simply remove them all together to save additional space. that way you save some code too. 
ie: just delete the whole line

JoshHawley, 2015-11-05:
Keep in mind that this library was not designed for the ATTiny, even on larger Arduinos, there are functions associated with this sensor that can not be implemented because of the memory restrictions. Also, to even get the serial debug info on the ATTiny is a pain. Do you even want it?

All that said, you can probably get it to work, but it will take chopping out some parts. Especially since you will want to save extra room for your other code.

gion86, 2015-11-07:
It is already working perfectly.. I had just to comment out some debug line in the library, but the whole functionality is there! I'm using it to open my garage :) Look at my repository...

The Attiny debug serial is also working! In the ATTiny core there is the *Tiny Debug serial*, which is at fixed speeds, and only transmitting, but it's enough for me!

gion86, 2015-11-07:
I was thinking at a pull request with some optimization to the library, but I might not have the time to do it :(

bboyho, 2020-05-19:
This issue looks stale. As JoshHawley stated, the library was written with the ATmega328P in mind so users would need to modify the library to get it working with another microcontroller. It sounds like you [got it working with an ATtiny](https://github.com/gion86/Fingerprint/blob/master/Documentation/Examples/GarageFinger_LCD_Menu.c). I'll be closing this issue out.

closed 2020-05-19 by bboyho