sparkfun/Fingerprint_Scanner-TTL #25: fps.GetEnrollCount() can only be called a limited amount of times

vladdeSV, 2016-06-13:
Calling `fps.GetEnrollCount()` a certain number of times hangs the program.

The following code freezes after 92 calls.
```c++
#include "FPS_GT511C3.h"
FPS_GT511C3 fps(2, 3);

void setup()
{
    Serial.begin(9600);
    fps.Open();
    fps.SetLED(true);
}

void loop()
{
    Serial.println(fps.GetEnrollCount());
}
```

*Tested on an Arduino Nano.*

JoshHawley, 2016-06-13:
I did some memory testing with this code before I posted it and it was not leaking. I was testing on an Uno, but I wouldn't think that should make too much difference? 

Is it always exactly 92 times? 
Try adding a 100ms delay in the loop to see if we are dealing with some kind of serial timing issue. 
Can you tell which line of code it breaks on? Does it always stop on the same line of code?

vladdeSV, 2016-06-14:
Adding a delay did have no direct effect. However, in my project with a bit more code the program would only call the function 62 times.

I'm not a C programmer, but if you look at the actual changes, i added `delete cp;` after the variable had been assigned something `new`.

Sent from my rymdtelefon

bboyho, 2018-03-15:
It looks like your fix has been merged [ [https://github.com/sparkfun/Fingerprint_Scanner-TTL/pull/26](https://github.com/sparkfun/Fingerprint_Scanner-TTL/pull/26) ]. I will be closing this issue. Thank you sharing the fix with the community! =)

closed 2018-03-15 by bboyho