sparkfun/Fingerprint_Scanner-TTL #27: Question: Arduino hardware tested & TTL Voltage Level

mikaelxu, 2016-06-24:
Hi, I have 2 questions.

1. Which Arduino have you tested this library with? Uno?
2. I read from Sparkfun that this module's UART TTL logic is 3.3V (https://www.sparkfun.com/products/11792). So which Arduino board to use is quite important in this case. Most Arduino boards have 5V TTL logic. Some have 3.3V TTL logic.

JoshHawley, 2016-07-20:
I tested this on an Uno. I used resistors to create a basic level shifter.

More info on my blog post: http://wordpress.hawleyhosting.com/ramblings/?p=375

bboyho, 2017-08-25:
Hi mikaelxu,

1. <p>Hawley's Arduino library was written for the Arduino Uno with the Atmega328P. The FPS_GT511C3 library may not work for all microcontrollers using the Arduino IDE. As you move away from the ATmega328P family, you may need to modify the code or port the library over to get it working. Theoretically, it should work with any Atmega328P bootloaded with an Arduino (i.e. [RedBoard Programmed with Arduino](https://www.sparkfun.com/products/13975) , [5V/16MHz Arduino Pro Mini](https://www.sparkfun.com/products/11113), [3.3V/8Mhz Arduino Pro Mini](https://www.sparkfun.com/products/11114), [Serial LCD Kit w/ Atmega328 bootloaded with Arduino](http://www.instructables.com/id/DIY-Fingerprint-Scanning-Garage-Door-Opener/), etc. ... ). At lower voltages, the microcontroller will clock at a lower speed but it should not have any issues affecting the overall functionality when using a baud rate of 9600. I personally have tested the example code on the RedBoard, [Arduino Mega 2560](https://www.sparkfun.com/products/11061), and [3.3V/8MHz Pro Micro](https://www.sparkfun.com/products/12587). There did not seem to be any issues but I did need to modify the software defined pins for the Atmega2560 and the Atmega32U4 based on the note in [Arduino.cc's software serial library](https://www.arduino.cc/en/Reference/softwareSerial).</p> <p>When testing it on the Arduino Due, there were issues compiling. You would need to modify the code and use it with the hardware serial UARTs because the Arduino Due does not support software serial. This old forum post will explain why there are compilation errors with the example code using the Arduino Due board definition => [Arduino.cc Forums: SoftwareSerial for Arduino Due](http://forum.arduino.cc/index.php?topic=142902.0). To my knowledge, there has not been a fix to get the software serial library working on the Arduino Due. </p>

2. The datasheets indicate that the fingerprint scanners use 3.3V logic. If you are using a 5V Arduino, you would need to use a [logic level converter](https://www.sparkfun.com/products/12009) or resistors for voltage division [ try looking at the Uni-Directional circuit design from our old logic level converter - [https://learn.sparkfun.com/tutorials/using-the-logic-level-converter#hardware-overview](https://learn.sparkfun.com/tutorials/using-the-logic-level-converter#hardware-overview) ] to safely connect to the fingerprint scanner. If you are using a 3.3V Arduino, you can just connect the UART pins together directly since the system is already 3.3V logic. To verify the Arduino's logic level, feel free to check with a multimeter or oscillscope.

closed 2017-08-25 by bboyho