sparkfun/Fingerprint_Scanner-TTL #28: fetch fingerprint database

felipefuller, 2016-07-11:
hi!

i'm working in a assistance control project. I need to fetch the template from the FPS's database through an arduino. Is there any way to do this? Because the GetTemplate() function is not created.

Thanks.

Alarde, 2016-07-11:
Following your question. As you already know, I'm very interested on it.

JoshHawley, 2016-07-14:
The problem is buffer space. The Arduino has almost no ram, so getting any real data from the FPS is problematic, and needs to be streamed somewhere else.

Alarde, 2016-07-14:
Uhm... 

Ok, let's supoused that we will work with Raspberry, here the scenario is a new one and now, we should have enough memory to cover the spectatives.

Is it right? Sorry if I'm telling stupid, I'm a bit new with this.

JoshHawley, 2016-07-16:
yeah the RPi has tons of memory. It should be fine with it, but it will require some reworking. I don't think the SoftwareSerial library works on RPi.

dannyd43, 2017-03-25:
Hello @JoshHawley the Arduino has 2048 bytes of ram as stated on the Arduino tutorial pages, the template is 498 bytes is there not enough memory to both get and set the templates? 

Source: https://www.arduino.cc/en/tutorial/memory

JoshHawley, 2017-03-25:
You are correct. However, I was not able to make it work. It has been a few years since I tried it, but I think it had something to do with the way the memory was buffered inside SoftwareSerial. iirc it stored it in too many places or something. I had toyed with the idea of using using a hardware serial of some kind, but never got around to it. Also, remember that there is no flow control coming from the FPS that also complicates things. I don't have much time to tinker with this stuff any more. If you do make it work please let me know (or maybe branch this code, then do a pull request). Many people here have asked for this.