sparkfun/Fingerprint_Scanner-TTL #30: Hardware serial

mattanyah, 2016-10-02:
hello 
how can I correct library for connect fps on hardware tx rx serial pins? my arduino board is nano and all digital pins are full. thanks for help

JoshHawley, 2017-01-18:
You would need to replace the SoftwareSerial library with another library.

TTKK2017, 2017-01-24:
I'm also having this issue I feel, do you know of any libraries that would be? If not, which ports would I use for the Arduino UNO, as none of my ports are full.

bboyho, 2017-08-28:
Hi mattanyah ,

I do not have the specific details on how to do this. I have not tried this before but you would probably need to go into the FPS_GT511C3 library and replace the instances of the software serial library with the Arduino.cc Serial library [ [https://www.arduino.cc/en/Reference/Serial](https://www.arduino.cc/en/Reference/Serial) ].

**Increasing I/O**

I personally would try to other alternatives. I am not sure what you are using for the digital pins. If you can, try using a multiplexer (a.k.a. mux) or I/O expander to increase the number of I/O pins so that you can use the defined software serial UART. SparkFun has a few options in the catalog [ [https://www.sparkfun.com/search/results?term=io+expander+mux](https://www.sparkfun.com/search/results?term=io+expander+mux) ]. You can also try to use an Arduino Mega 2560 [ [https://www.sparkfun.com/products/11061](https://www.sparkfun.com/products/11061) ] or a 5V tolerant Teensy w/ the Arduino add-on [ [https://www.sparkfun.com/products/13736](https://www.sparkfun.com/products/13736) ]. Just make sure to redefine the software serial pins and rewire the circuit depending on the microcontroller that you use.

While redundant, maybe you can add an additional Arduino microcontroller dedicated to the fingerprint scanner. You would need to write code to communicate between the two Arduinos but it can be as simple as checking the state of a pin if the fingerprint is scanning or identifying a fingerprint.

bboyho, 2017-08-28:
Hi @TTKK2017 ,

I do not have the specific details on how to do this. I have not tried this before but you would probably need to go into the FPS_GT511C3 library and replace the instances of the software serial library with the Arduino.cc Serial library [ [https://www.arduino.cc/en/Reference/Serial](https://www.arduino.cc/en/Reference/Serial) ].

I assume that the I/O pins are free since your post indicates that the none of the ports are full. I recommend using the defined software serial pins on the FPS_GT511C3 library. If pins 4 & 5 are reserved, I recommend redefining the software serial pins on your Arduino Uno's other I/O pins [ [https://www.arduino.cc/en/Reference/softwareSerial](https://www.arduino.cc/en/Reference/softwareSerial) ].

**Resources and Going Further**

Otherwise, this is a great question to check or post in the Arduino.cc Forums [ [http://forum.arduino.cc/](http://forum.arduino.cc/) ].

closed 2017-08-28 by bboyho