sparkfun/Fingerprint_Scanner-TTL #31: about getImage and getTamplate function

chamaraManoj, 2016-10-07:
I tried to implement the getImage function. However in my serial monitor in arduino I got some data. Using that data I tried to develop an finger print image. But completely I got weird image. I know that due to memory restriction it cannot be take the full image. But using the received data can I develop and image to a certain extent. 

Here with I have attached the matlab code I implemented and varible 'data' has character array of 62448 which I took from the data packet received.

[matlab.zip](https://github.com/sparkfun/Fingerprint_Scanner-TTL/files/515904/matlab.zip)

eron93br, 2017-03-27:
Hello, I'm trying to get the image from through arduino serial.

Could you share your code for the getImage and getTemplate functions?

bboyho, 2020-05-20:
Hi,

This looks a bit old and it sounds like you are looking for technical assistance; the GitHub issues are used primarily for bugs and pull requests related to the libraries. I am not exactly sure what is going on with the image since the functions are not implemented with Arduino. It's possible that there might be a limitation when using software serial and passing it to the hardware. 

If you have not already, you may want to post in the [SparkFun forums](https://forum.sparkfun.com/index.php) or [Arduino forums](https://forum.arduino.cc/) to see if anyone can provide some insight.

closed 2020-05-20 by bboyho