sparkfun/Fingerprint_Scanner-TTL #32: Library won't compile on ESP8266 family boards

assaf64, 2016-12-25:
The EPS8266 boards, in particular the WeMos D1 mini development boards, are becoming popular because of their WiFi support (for IoT). However these boards are provided with a lighter version of "SoftwareSerial" that does not support methods such as "end" and "listen".

The FPS library uses these methods in two locations.  In function "ChangeBoudRate" the "end" method is used, and in function "GetResponse" the "listen" method is used.

Is there a workaround that avoid the use of these two methods?

Please advise.

ryonlabaw, 2017-01-14:
I have been using Wemos D1 Mini + sparkfun fingerprint scanner without issue.   My arduino IDE is using the ESP8266 Arduino core development version from github, not the downloadable/release version.  Maybe that is the reason I am not seeing any compile errors.

Also, the end + listen commands don't look very important the way they are used. I bet you can comment those out.

cdiaz, 2017-10-26:
@assaf64 I have the same problem, have you solved it?

assaf64, 2017-10-26:
I remember having some minor issues (perhaps few compilation warnings), but
it worked with no problems.

xnfctn, 2018-01-27:
@ryonlabaw can you please have a look at my thread   #49 ? i didnt get any compile error, but communication problems. how did you wired both? is my wireing correct?

bboyho, 2020-05-19:
This looks like a stale issue and I will be closing this out.

I'm not sure if you are still having problems but testing it out with the SparkFun ESP8266 Thing Dev on pins 4 (SS Rx) and 5 (SS Tx), it seems to be working fine to enroll and ID a finger. You may want to download the latest ESP8266 board add-on (seems to be working for v2.6.3. The ESP8266 community has been constantly updating the core and the updates seem to have resolved a few issues with using Software Serial on an ESP8266.

closed 2020-05-19 by bboyho