sparkfun/Fingerprint_Scanner-TTL #33: Multi-User Help

rohinrohin, 2017-01-05:
Is there a way to add support for more than 20 users, for example, generate a fingerprint "hash" or a template string that I can match with a dB, or even on a pi with some python code?

tldr; is it possible to support more than 20 users with this module? If so, how. 

Any help is appreciated. 

And thanks for the code, bought it and got it working right out of the box!

JoshHawley, 2017-01-18:
The GT-511C3 supports 200 (not 20) fingerprints. Do you really need more than that?

In theory, you can download the "templates" from the device, but I was unable to do it with an arduino because of memory restrictions. While I have not personally done it, I did receive an email from someone else that was able to replace SoftwareSerial with another library, and use it on a RPi. It would have much more memory to work with (and possibly a hardware buffer too)

rohinrohin, 2017-01-31:
@JoshHawley Yes, I am building a prototype for a finger print based attendance system for a university. In this case, there are more than 200 people. However, I did take a look at downloading the templates from the device into an rPi. 

But, since the "templates" do not follow any common format, the templates have to be flashed back into the flash to be verified. Is there any other way where I can match the templates without flashing it back to the device.

JoshHawley, 2017-01-31:
If you have downloaded the templates from the device, you should be able to do a byte by byte comparison of them to check for equality. Its pretty tedious though. It may be that this is the wrong piece of hardware for you. It seems too self contained for what you are trying to accomplish.

rohinrohin, 2017-02-09:
Alright, thanks for your help.

closed 2017-02-09 by rohinrohin