sparkfun/Fingerprint_Scanner-TTL #34: Issues when compilling

jamesyesjames, 2017-01-17:
Hi,
I'm new to Ardunio but have a background in programing.
I've connected the FPS as per diagrams to pins 5 & 4 to a Uno but when I upload the program I get the following errors from the IDE. Am i doing something really silly??
Any help would be amazing, thank you 

Arduino: 1.6.13 (Mac OS X), Board: "Arduino/Genuino Uno"

In file included from /Users/jamesbernhard/Documents/Arduino/Fingerprint_Scanner-TTL-master/FPS_GT511C3/Examples/FPS_Enroll/FPS_Enroll.ino:10:0:
sketch/FPS_GT511C3.h:12:21: warning: extra tokens at end of #include directive
 #include "Arduino.h";
                     ^
sketch/FPS_GT511C3.h:13:28: warning: extra tokens at end of #include directive
 #include "SoftwareSerial.h";
                            ^
/var/folders/v9/wb35yrt521vdgzvz8s0dyvjr0000gn/T//cc4oNGo0.ltrans0.ltrans.o: In function `__static_initialization_and_destruction_0':
/Users/jamesbernhard/Documents/Arduino/Fingerprint_Scanner-TTL-master/FPS_GT511C3/Examples/FPS_Enroll/FPS_Enroll.ino:18: undefined reference to `FPS_GT511C3::FPS_GT511C3(unsigned char, unsigned char)'
/var/folders/v9/wb35yrt521vdgzvz8s0dyvjr0000gn/T//cc4oNGo0.ltrans0.ltrans.o: In function `setup':
/Users/jamesbernhard/Documents/Arduino/Fingerprint_Scanner-TTL-master/FPS_GT511C3/Examples/FPS_Enroll/FPS_Enroll.ino:24: undefined reference to `FPS_GT511C3::Open()'
/Users/jamesbernhard/Documents/Arduino/Fingerprint_Scanner-TTL-master/FPS_GT511C3/Examples/FPS_Enroll/FPS_Enroll.ino:25: undefined reference to `FPS_GT511C3::SetLED(bool)'
/var/folders/v9/wb35yrt521vdgzvz8s0dyvjr0000gn/T//cc4oNGo0.ltrans0.ltrans.o: In function `Enroll':
/Users/jamesbernhard/Documents/Arduino/Fingerprint_Scanner-TTL-master/FPS_GT511C3/Examples/FPS_Enroll/FPS_Enroll.ino:40: undefined reference to `FPS_GT511C3::CheckEnrolled(int)'
/Users/jamesbernhard/Documents/Arduino/Fingerprint_Scanner-TTL-master/FPS_GT511C3/Examples/FPS_Enroll/FPS_Enroll.ino:43: undefined reference to `FPS_GT511C3::EnrollStart(int)'
/Users/jamesbernhard/Documents/Arduino/Fingerprint_Scanner-TTL-master/FPS_GT511C3/Examples/FPS_Enroll/FPS_Enroll.ino:48: undefined reference to `FPS_GT511C3::IsPressFinger()'
/Users/jamesbernhard/Documents/Arduino/Fingerprint_Scanner-TTL-master/FPS_GT511C3/Examples/FPS_Enroll/FPS_Enroll.ino:49: undefined reference to `FPS_GT511C3::CaptureFinger(bool)'
/Users/jamesbernhard/Documents/Arduino/Fingerprint_Scanner-TTL-master/FPS_GT511C3/Examples/FPS_Enroll/FPS_Enroll.ino:54: undefined reference to `FPS_GT511C3::Enroll1()'
/Users/jamesbernhard/Documents/Arduino/Fingerprint_Scanner-TTL-master/FPS_GT511C3/Examples/FPS_Enroll/FPS_Enroll.ino:55: undefined reference to `FPS_GT511C3::IsPressFinger()'
/Users/jamesbernhard/Documents/Arduino/Fingerprint_Scanner-TTL-master/FPS_GT511C3/Examples/FPS_Enroll/FPS_Enroll.ino:57: undefined reference to `FPS_GT511C3::IsPressFinger()'
/Users/jamesbernhard/Documents/Arduino/Fingerprint_Scanner-TTL-master/FPS_GT511C3/Examples/FPS_Enroll/FPS_Enroll.ino:58: undefined reference to `FPS_GT511C3::CaptureFinger(bool)'
/Users/jamesbernhard/Documents/Arduino/Fingerprint_Scanner-TTL-master/FPS_GT511C3/Examples/FPS_Enroll/FPS_Enroll.ino:62: undefined reference to `FPS_GT511C3::Enroll2()'
/Users/jamesbernhard/Documents/Arduino/Fingerprint_Scanner-TTL-master/FPS_GT511C3/Examples/FPS_Enroll/FPS_Enroll.ino:63: undefined reference to `FPS_GT511C3::IsPressFinger()'
/Users/jamesbernhard/Documents/Arduino/Fingerprint_Scanner-TTL-master/FPS_GT511C3/Examples/FPS_Enroll/FPS_Enroll.ino:65: undefined reference to `FPS_GT511C3::IsPressFinger()'
/Users/jamesbernhard/Documents/Arduino/Fingerprint_Scanner-TTL-master/FPS_GT511C3/Examples/FPS_Enroll/FPS_Enroll.ino:66: undefined reference to `FPS_GT511C3::CaptureFinger(bool)'
/Users/jamesbernhard/Documents/Arduino/Fingerprint_Scanner-TTL-master/FPS_GT511C3/Examples/FPS_Enroll/FPS_Enroll.ino:70: undefined reference to `FPS_GT511C3::Enroll3()'
/var/folders/v9/wb35yrt521vdgzvz8s0dyvjr0000gn/T//cc4oNGo0.ltrans0.ltrans.o: In function `__static_initialization_and_destruction_0':
/Users/jamesbernhard/Documents/Arduino/Fingerprint_Scanner-TTL-master/FPS_GT511C3/Examples/FPS_Enroll/FPS_Enroll.ino:18: undefined reference to `FPS_GT511C3::~FPS_GT511C3()'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

jamesyesjames, 2017-01-17:
Please forget my comment! turns out I had a faulty cable!

JoshHawley, 2017-01-18:
I'm glad you got it working =)

jamesyesjames, 2017-01-20:
Thanks! 

Just wondering if I can  ask your advise on one small matter?
I have attached the FPS to a fairly long ribbon cable (15f with the Resistors at the Ardunio side) and the senor is much slower at detecting a correct print. Should I change the two resistors? If so what would you recommend?  

Many, many, thanks
James

JoshHawley, 2017-01-20:
Whenever you add length to a cable, you are adding the resistance of the cable itself. I would check that you are actually getting the proper voltages out the far end. I would disconnect the FPS, and just plug the arduino side of the voltage divider into the 5v bus, then use a volt meter on the FPS side to see that if you are getting the proper 3.3v. if necessary, adjust your resistor values accordingly.

JoshHawley, 2017-01-20:
Also, since the Arduino side is higher voltage (5v) than the FPS side (3.3v), I would put the resistors on the FPS side.

jamesyesjames, 2017-01-20:
Hi Josh,

Thanks for that - Is it just the 680R (between RX and TX) I need to change or both resistors?
Sorry a bit out of practice with this stuff! 
I will do as you advised and hook up a meter and try a few different options...

Thank you again,
James

JoshHawley, 2017-01-31:
It will depend on the readings from the volt meter. You will need to measure the actual voltage, and then do the math to figure out which resistors you need. 
http://www.raltron.com/cust/tools/voltage_divider.asp

bboyho, 2017-08-25:
Hi jamesyesjames,

It looks like you got the code to compile.

As for the voltage divider, I agree with Hawley. 15 feet is a bit long between the fingerprint scanner and Arduino. There a few methods of approaching this problem. The quickest may be to adjust the resistors based on your voltage so that the devices are able to read a logic HIGH [ https://learn.sparkfun.com/tutorials/logic-levels#33-v-cmos-logic-levels ].

I remember doing some stress tests (with the original 560 Ohm and 1000Ohm resistor) for someone  using the GT511C1R and measuring the voltage with a multimeter, I noticed that the divided voltage was about 3.14V. When the GT511C1R was connected, the voltage dropped to about 2.97V. This is probably not high enough for the fingerprint scanner to recognize a logic HIGH reliably.  I was able to get it working when using a voltage divider (with 10kΩ and 20kΩ) similar to the setup used in our previous logic level converter’s application circuit [try looking at the Uni-Directional circuit => https://cdn.sparkfun.com/assets/b/0/e/1/0/522637c6757b7f2b228b4568.png ]. Testing the circuit with the resistors replaced, my multimeter read a divided voltage of 3.3V. When the GT511C1R was connected, the voltage dropped to about ~3.18V-3.04V. This was sufficient enough for the fingerprint scanner to receive the commands from an Arduino. I tested this with the blink examples and there were no problems with my 6" M/M jumper wire.

Otherwise, you might want to consider trying a dedicated bi-directional logic level converter [ https://www.sparkfun.com/products/12009 ] or a converter that is able to send serial data over long distances.

closed 2017-08-25 by bboyho

kishupatel, 2017-09-22:
Please quickly help me to resolve this error

In file included from C:\Users\LENOVO\Desktop\FPS_Blink_1\FPS_Blink_1.ino:1:0:

: warning: extra tokens at end of #include directive

 #include "Arduino.h";

                    : warning: extra tokens at end of #include directive

 #include "SoftwareSerial.h";