sparkfun/Fingerprint_Scanner-TTL #35: Error in response packet and SerialDebug

TTKK2017, 2017-01-20:
`C:\Users\Administrator\Documents\Arduino\libraries\FPS_GT511C3\FPS_GT511C3.cpp: In constructor 'Response_Packet::Response_Packet(byte*, bool)':

C:\Users\Administrator\Documents\Arduino\libraries\FPS_GT511C3\FPS_GT511C3.cpp:89:108: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

  CheckParsing(buffer[0], COMMAND_START_CODE_1, COMMAND_START_CODE_1, "COMMAND_START_CODE_1", UseSerialDebug);`

I'm using an Arduino Uno on Win 7 at my school, but I have access to admin rights. I'm brand new to Arduino, but I wrote a few lines for simple commands that have compiled in the past, but this Serial issue keeps coming up and the FPS didn't seem to be communicating either. Has anyone else seen these or know how to fix any errors?

JoshHawley, 2017-01-31:
It looks like there is something wrong with the compiler. Did the install go ok?
The warning about the conversion shouldn't stop anything from working.

TTKK2017, 2017-02-06:
Just reinstalled IDE, still getting the error but you're saying tat shouldn't stop it from working. Is there an easy way to determine if the FPS is fried? I kinda dived right in for a class with my first Arduino project :/

TTKK2017, 2017-02-16:
Got it all working after a wiring switch around, thanks for the help  and awesome library! :3

closed 2017-02-16 by TTKK2017