sparkfun/Fingerprint_Scanner-TTL #37: Interfacing with Arduino ProMicro (3.3V ATmega32U4)

Alpha-Amol, 2017-02-14:
Hi,
I have connected the GT-511C1R bought from sparkfun to the ProMicro board and have used the Sparkfun FPS_GT511C3 - Arduino Library to interface with the sensor.
Connection as follows:-
//    Hardware setup - FPS connected to Arduino ProMicro (3.3V ATmega32U4):
//    digital pin 10(arduino rx, fps tx)
//    digital pin 11(arduino tx, fps rx)
//    Arduino 5V to Vin of FPS sensor
//    Arduino Ground to Gnd of FPS sensor

`
#include "FPS_GT511C3.h"
#include "SoftwareSerial.h"

// Hardware setup - FPS connected to:
//    digital pin 10(arduino rx, fps tx)
//    digital pin 11(arduino tx, fps rx)
//    Arduino 5V to Vin of FPS sensor
//    Arduino Ground to Gnd of FPS sensor

FPS_GT511C3 fps(10, 11); // RX, TX

void setup()
{
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
  Serial.println("Started the setup function!!");
  fps.UseSerialDebug = true; // so you can see the messages in the serial debug screen
  fps.Open();
}

void loop()
{
  // FPS Blink LED Test
  fps.SetLED(true); // turn on the LED inside the fps
  delay(1000);
  fps.SetLED(false);// turn off the LED inside the fps
  delay(1000);
}
`

The only output on my serial monitor is this:
`
Started the setup function!!
FPS - Open Started
FPS - SEND: "55 AA 01 00 00 00 00 00 01 00 01 01"
`

It seems that the only executes the first line of the code and does not go any further. The baud rate for interfacing FPS is 9600 but i changed it to 115200 to check if it works but it does not!!

JoshHawley, 2017-02-14:
To change the baud rate, you need to boot and connect to it at 9600, then use the ChangeBaudRate command to get to the faster speed. 

Try feeding the FPS 3.3v on its vcc pin instead of 5v. The C3 is supposed to support 3.3v - 6v on the pin, but I had issues with it on 5v.

TTKK2017, 2017-02-16:
Not a question here, but just wanted to thank OP for posting that pic of the pin layouts on the FPS. I couldn't find that anywhere, and it was so helpful!

Alpha-Amol, 2017-02-16:
Thanks JoshHawley for your help and the library!!
And TTKK2017 use 3.3V for power supply and default baud rate for FPS connection is 9600!!
[image]

bboyho, 2017-08-25:
Hi Alpha-Amol, 

**Basic Troubleshooting**

For basic troubleshooting, I recommend checking your connections. The 3.3V/8MHz Pro Micro does not have headers installed. If you are using header pins, make sure that it is sufficiently [soldered](https://learn.sparkfun.com/tutorials/how-to-solder-through-hole-soldering). Assuming that you are using water soluble flux, make sure to remove any flux residue that was left on the board. This can cause temporary shorts due to dendrites and interfere with the serial connection. If you are using the [JST SH Jumper 4 Wire Assembly](https://www.sparkfun.com/products/10359), the cable’s wire is relatively thin compared to a breadboard's socket. A small bump can break the connection requiring you to power cycle the fingerprint scanner or reset the Arduino. Also, make sure that Tx is connected to Rx and Rx is connected to Tx. This may sound trivial but I have seen a lot of customers with these problems.

**Voltage Division**

Since you are using the [3.3V/8MHz Pro Micro](https://www.sparkfun.com/products/12587), the logic level should be 3.3V. You would not need any voltage dividers between the fingerprint scanner and the 3.3V Pro Micro. 

**Software Serial Pin Definition**

The FPS_GT511C3 library may not work for all microcontrollers using the Arduino IDE. As you move away from the ATmega328P family, you may need to modify the code or port the library over to get it working.

I'm not sure where you are connecting the software defined Tx pin. The [3.3V/8MHz Pro Micro](https://learn.sparkfun.com/tutorials/pro-micro--fio-v3-hookup-guide#hardware-overview-pro-micro) does not have a pin 11 broken out on the PCB. Try redefining the software serial pins:

    FPS_GT511C3 fps(10, 11); // RX, TX

to

     FPS_GT511C3 fps(10, 16); // (Arduino SS_RX = pin 10, Arduino SS_TX = pin 16)

Doing a quick test with a known good GT-511C1R, 3.3V/8MHZ Pro Micro, and no resistors attached, I was able to get the _FPS_Blink.ino_ and FPS_IDFinger.ino_ to work without any problems.

closed 2017-08-25 by bboyho