sparkfun/Fingerprint_Scanner-TTL #42: Fail to save after three enrollment

cooldog1314, 2017-07-08:
I am using a raspberry pi to drive the scanner. I follow the order, Open->StartEnroll->Capture->Enroll1->Capture->Enroll2->Capture->Enroll3, to enroll my fingerprint. I get ACK for all commands but when I count my enrollments/ check the ID occupancy, the scanner sends me an Error and tells me its database is empty. What could be the reason?

JoshHawley, 2017-07-08:
Have you verified that you are trying to save to an empty id?

cooldog1314, 2017-07-08:
Before EnrollStart, I use CheckEnrolled command to verify id's availability. It gives me error code 0x1004 which means the id is not use.

JoshHawley, 2017-07-08:
What ID number are you trying to use?

JoshHawley, 2017-07-08:
Could you post the snippet of code where you call the FPS?

cooldog1314, 2017-07-08:
I have tried different ID such as 0, 1, 2, 24 and 199. All test encountered the same problem. My code is based on library provided by github user zafartahirov. The link is https://github.com/zafartahirov/fingerpi

The enrolment code is attached below:
```
#!/usr/bin/env python
import time
import fingerpi as fp
# from fingerpi import base

# import struct
import time
# import matplotlib.pyplot as plt
import pickle

def printByteArray(arr):
    return map(hex, list(arr))

f = fp.FingerPi()

print 'Opening connection...'
f.Open(extra_info = True, check_baudrate = True)

#print 'Changing baudrate...'
#f.ChangeBaudrate(9600)

enrollid = 199;
response = f.CheckEnrolled(enrollid)
if response[0]['ACK']:
    print 'ID ',enrollid,'is not occupied'
else:
    print 'Unknown Error occured', response[0]['Parameter']

    
f.CmosLed(True)
response = f.EnrollStart(enrollid)
if response[0]['ACK']:
    print'enroll started'
else:
    print 'ENROLL Start Error occured', response[0]['Parameter']
# print f.UsbInternalCheck()

# enroll
print 'press finger to enroll ID',enrollid

print 'Place the finger on the scanner'
while True:
    response = f.IsPressFinger()
    if response[0]['ACK']:
        if response[0]['Parameter']==0:
            print'finger pressed'
            break
    time.sleep(1)

    # response = f.IsPressFinger()
response = f.CaptureFinger()
if response[0]['ACK']:
    #if response[0]['Parameter'] != 'NACK_FINGER_IS_NOT_PRESSED':
        #print 'Unknown Error occured', response[0]['Parameter']    
    print 'Image captured! Please remove finger'
    response = f.Enroll1()
    if response[0]['ACK']:
        print 'ENROLL #1 SUCCESSFUL'
    else:
        print 'ENROLL #1 Error occured', response[0]['Parameter']    
    while True:
        response = f.IsPressFinger()
        if response[0]['ACK']:
            if response[0]['Parameter']!=0:
                print'finger removed'
                break
        time.sleep(1)    
    print 'place your finger for the second enrollment'
    while True:
        response = f.IsPressFinger()
        if response[0]['ACK']:
            if response[0]['Parameter']==0:
                print'finger pressed'
                break
        time.sleep(1)
    response = f.CaptureFinger()
    if response[0]['ACK']:
        #if response[0]['Parameter'] != 'NACK_FINGER_IS_NOT_PRESSED':
        #print 'Unknown Error occured', response[0]['Parameter']    
        print 'Image captured! Please remove finger'
        response = f.Enroll2()
        if response[0]['ACK']:
            print 'ENROLL #2 SUCCESSFUL'
        else:
            print 'ENROLL #2 Error occured', response[0]['Parameter']
        while True:
            response = f.IsPressFinger()
            if response[0]['ACK']:
                if response[0]['Parameter']!=0:
                    print'finger removed'
                    break
        time.sleep(1)
        print 'place your finger for the Third enrollment'
        while True:
            response = f.IsPressFinger()
            if response[0]['ACK']:
                if response[0]['Parameter']==0:
                    print'finger pressed'
                    break
            time.sleep(1)
        response = f.CaptureFinger()
        if response[0]['ACK']:
            print 'Image captured! Please remove finger'
            response = f.Enroll3()
            if response[0]['ACK']:
                print 'ENROLL #3 SUCCESSFUL'
            else:
                print 'ENROLL #3 Error occured', response[0]['Parameter']
        else:
            print'capture 3 failed', response[0]['Parameter']
    else:
        print'capture 2 failed', response[0]['Parameter']
else:
    print'capture 1 failed', response[0]['Parameter']

    
while True:
    response = f.GetEnrollCount()
    if response[0]['ACK']:
        print 'Total ID Enrolled is ',response[0]['Parameter']
        break
    time.sleep(1)
    print 'Unknown Error occured #10', response[0]['Parameter']

  
```
[image]

cooldog1314, 2017-07-11:
any guess for its reason? do you have any suggestion on debug?

bboyho, 2018-03-15:
Hi,

I have not used the fingerprint scanner beyond the Hawley's Arduino library. I'm not exactly sure what the error code in Python means or why the scanner did not save the ID after enrolling. If you have not resolved the issue, you may want to look at this example code for the fingerprint scanner used in a different repository => [ [https://github.com/QuickGroup/pyGT511C3](https://github.com/QuickGroup/pyGT511C3) ]. It was from a Raspberry Pi forum and it seemed to work for the author [ [https://www.raspberrypi.org/forums/viewtopic.php?p=636148&sid=bf26e4f230b44ac7fe21d404fc2cd7f8#p636148](https://www.raspberrypi.org/forums/viewtopic.php?p=636148&sid=bf26e4f230b44ac7fe21d404fc2cd7f8#p636148) ]

closed 2020-05-18 by bboyho