sparkfun/Fingerprint_Scanner-TTL #48: Raspberry Pi 3 with Windows 10 Iot

bahadiracar, 2017-11-23:
Hi,

I want to run the GT-511C3 module with Raspberry Pi 3. (C #)
Are there examples of "Enroll" and "IDFinger" on this subject?

Please help me...

Regards

JoshHawley, 2018-01-26:
Not that I know of. The full protocol spec is in the pdf file though if you want to write one.

bboyho, 2020-05-19:
This issue is stale and I will be closing this out. The fingerprint scanners have been tested with Hawley's Arduino library. Unfortunately, we have not heard any specific examples that use the fingerprint scanner and the Windows IoT core. While redundant, your best bet may be to have the Arduino do a majority of the work and connect it to the Pi if you have not already.

closed 2020-05-19 by bboyho