sparkfun/Fingerprint_Scanner-TTL #49: Communication Problem with Wemos d1 mini pro

xnfctn, 2018-01-26:
Hi all,

Problem: I didnt get the FPS_Blink.ino to work.

i have a GT511-C1R and Wemos D1 mini pro. I wired both components like in the attached picture.

[image]

As the wemos IOs runs @3.3v, so there is no need for any kind of voltage division as i understand.

i got follwing output in the Serial Monitor 
#################################################
Soft WDT reset

ctx: cont 
sp: 3ffefd80 end: 3ffeff90 offset: 01b0

>>>stack>>>
3ffeff30:  0000000c 3ffeec28 00000055 40202493  
3ffeff40:  3fffdad0 00000001 3ffeeea8 3ffeef60  
3ffeff50:  3fffdad0 3fff0e94 3ffeec28 40202564  
3ffeff60:  feefeffe 00000000 3ffeef58 4020205e  
3ffeff70:  feefeffe feefeffe feefeffe 40202f74  
3ffeff80:  feefeffe feefeffe 3ffeef70 40100710  
<<<stack<<<
⸮$⸮L⸮<⸮1⸮=⸮FPS - Open
FPS - SEND: "55 AA 01 00 00 00 00 00 01 00 01 01"
 ⸮n?⸮4⸮!⸮⸮⸮⸮OCAC⸮⸮FPS - Open
FPS - SEND: "55 AA 01 00 00 00 00 00 01 00 01 01"
##########################################################################

Did i something wrong? Any ideas what to test?

JoshHawley, 2018-01-26:
I am unfamiliar with that board's pinouts. Are you sure that you should not be using D4 and D5? Or else change the startup code to match the different pins. 

You are correct that no voltage divider is needed. 

The output only shows transmits, so the communications pins are suspect.

xnfctn, 2018-01-26:
Hi Josh,
following link shows the pinout of the wemos
https://wiki.wemos.cc/products:d1:d1_mini_pro

due to that IO pin4=d2 and IO pin5=d1

bboyho, 2018-01-27:
I as well am unfamiliar with the ESP8266 on the WeMos D1 Mini Pro.

**Wrong Pin Out with ESP8266 Arduino Core?**

Hmmm, it does look like someone has gotten the ESP8266 to work with the fingerprint scanner => [ [https://github.com/sparkfun/Fingerprint_Scanner-TTL/issues/32#issuecomment-339634009](https://github.com/sparkfun/Fingerprint_Scanner-TTL/issues/32#issuecomment-339634009) ]. I understand that the wiki states that "GPIO4 = D2" and "GPIO=D1". Try connecting the serial pins to software serial pins D4 and D5 or using a different set of pins. Since it is a 3.3V system, you do not need a logic converter/resistors for voltage division.

There may be a limitation with the Software Serial library on the ESP8266. I know that when you move from away from the ATmega328P-based microcontroller, you may need to redefine and redefine pins [ [https://www.arduino.cc/en/Reference/SoftwareSerial](https://www.arduino.cc/en/Reference/SoftwareSerial) ].

**Solder Joints**

Looking at your image, it looks like you did not solder to the board. You should make sure that your solder joints are connected and ideal. The fingerprint scanner may be getting power but it might not be getting the correct packets if the serial pins are not secure.

For more information, check out this section that I wrote for troubleshooting connections => [ [https://learn.sparkfun.com/tutorials/sparkfun-troubleshooting-tips#bad-solder-joints](https://learn.sparkfun.com/tutorials/sparkfun-troubleshooting-tips#bad-solder-joints) ].

xnfctn, 2018-01-27:
Thank you, I will Check the points...

ryonlabaw, 2018-01-27:
Its been a while since I messed with that scanner ... but I remember that it needed a good power supply or the scanner would reboot at random times. 

I didn't provide power via the wemos, so that is the first place I would check. 

Otherwise, its standard RX/TX ... I used software serial and github version of arduino-esp code ...

xnfctn, 2018-02-21:
I tryed with UNO and MEGA, both worked. So im sure it's a Software compatibility Problem with The esp8266

avirup171, 2018-05-10:
It's an issur with the watch dog timer. I have  been facing this too.
Possible workaround is
1: Adding delay before and after fps.open() of 100 ms seems to stabilise a bit but not fully.
I am also trying to figure out a way to solve this permenently but no leads as of now.

bboyho, 2020-05-19:
This looks like a stale issue and I will be closing this out.

I'm not sure if you are still having problems but testing it out with the SparkFun ESP8266 Thing Dev on pins 4 (SS Rx) and 5 (SS Tx), it seems to be working fine to enroll and ID a finger. You may want to download the latest ESP8266 board add-on (seems to be working for v2.6.3. The ESP8266 community has been constantly updating the core and the updates seem to have resolved a few issues with using Software Serial on an ESP8266.

closed 2020-05-19 by bboyho