sparkfun/Fingerprint_Scanner-TTL #57: Code doesnt work with ESP32 or ESP8266

philnicholson2005, 2018-11-11:
Josh

i am sure you're a busy man and im not sure that you are still working with this?

i would like to use this library with an ESP32 but unfortunately it will not compile with any ESP board selected? 

any ideas where im going wrong?

thanks Phil

fourstix, 2018-11-15:
Hi Phil,
Are you getting compile errors something like: class 'SoftwareSerial' has no member named 'end' ?  

I had some errors like that when compiling for the ESP8266.  They were caused by an older version of espSoftwareSerial in the ESP8266 board libraries.  I got the code to compile by updating the Arduino IDE to the latest version, 1.8.7 and then using the Board Manager to update the ESP8266 community libraries to their latest version 2.4.2.

After doing the updates, the libraries would compile clean using the Sparkfun ESP8266 Thing Dev board.  I think it would probably compile clean for any ESP8266 after updating the board libraries.

Good luck,
Gaston

philnicholson2005, 2018-11-15:
Gaston thanks!

Just tried it and thanks it does work for an ESP8266 so I can use them but ideally I wanted to use an ESP32

Regards

Phil

philnicholson2005, 2018-11-17:
Gaston

If I switched to an ESP8266 which pins would you use for this fingerprint scanner?

fourstix, 2018-11-18:
Hi Phil,
I think any unused I/O pins on the ESP8266  would be fine.  
Here's an example from the espSoftwareSerial library using pins 12 and 14. 
https://github.com/plerup/espsoftwareserial/blob/master/examples/swsertest/swsertest.ino

Good luck,
Gaston

PawelMiera, 2021-01-20:
I got it working on ESP32 :) 
You need to use HardwareSerial Serial2 instead of Software Serial. Here is the link to changed library:
https://github.com/PawelMiera/sparkfun-Fingerprint_Scanner-TTL-for-esp32

Good luck.