sparkfun/Fingerprint_Scanner-TTL #61: Library for GT-521F32

sivabharaniii, 2020-01-09:
Hi,
    I am working with GT-521F32,in that librarie some functions are not implemented due to memory restrictions on arduino.
I need those libraries to be implemented,I am using Arduino mega 2560 with in built wifi board which has more memory than arduino uno.
Functions to be implemented in libarary
1) GET IMAGE()
2) GET RAW IMAGE()
3) GET TEMPLATE ()
4)SET TEMPLATE()

brianrho, 2020-03-09:
My [library](https://github.com/brianrho/GT5X) implements these already if you're still interested.

bboyho, 2020-05-19:
Hi,

It looks like your question has been answered so I will be closing this out. There is a [pull request](https://github.com/sparkfun/Fingerprint_Scanner-TTL/pull/54) that implements those features and it looks like JoshHawley briefly reviewed it. However, I have not tested it out. It looks like brianrho has implemented those features with additional examples. ;)

closed 2020-05-19 by bboyho